Ultimaker/Cura #11210: Arachne: No infill generated in small cross sections when using extra infill walls

neutebach, 2022-01-06:
### Application Version

Arachne Beta 2

### Platform

Windows 11

### Printer

CFFFP

### Reproduction steps

To receive strong parts, more than the standard 2 wall lines may be chosen to increase the wall thickness. The print speed can only be chosen for outer and inner walls, but stays the same no matter how many inner walls there are, adding print time with every further inner wall. I chose to instead add extra infill walls that are then printed at the speed of the infill, allowing for the best compromise between strong parts and print time.

--> instead of a _Wall Line Count_ of 4, only 2 Walls but also 2 _Extra Infill Walls_ are selected.

[image]

### Actual results

For some small areas, presumably where the infill area is too small to accomodate all perimeters and **extra infill wall**s, there is **no infill generated** at all, when the above mentioned settings are in place (2 perimeters/walls, 2 extra infill walls).

[image]

[image]

Infill is however generated in these areas when **no extra infill walls** are selected:

[image]

### Expected results

Similar to what is output when **4 walls, but no extra infill walls** are selected (see images below), I would expect the extra infill walls to be placed inside the perimeters so that the area is filled. When the area cannot accomodate all extra infill walls, only the necessary amount of walls should be drawn, instead of the infill being omitted completely.

[image]

[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[arachne-demo-project.zip](https://github.com/Ultimaker/Cura/files/7824644/arachne-demo-project.zip)

fvrmr, 2022-01-07:
Hi @neutebach thank you for your bug report.
I will bring this up with the team for discussion. Keep you posted!

neutebach, 2022-01-07:
thank you @fvrmr 

I've just found that in some cases, infill with extra walls might be created after all when there is at least one connected area where all infill walls and the infill structure can be created - however this might just be coincidental. Nevertheless I tought I'd share it.

[image]

neutebach, 2022-01-07:
FWIW after looking at these screenshots again, this encapsules the whole beauty of how arachne is working, it's really amazing when you think back to how the old engine would handle these cross sections. I'm really excited for Cura 5, keep up the great work!

fvrmr, 2022-01-11:
I have discussed it with the team and turns out we already got a ticket related to this. I will link this issue to the ticket on our backlog. 
Devs see CURA-8067

closed 2022-02-15 by nallath