Ultimaker/Cura #11211: Multiplying Objets

MiguelEro, 2022-01-06:
### Application Version

4.11.0

### Platform

mac

### Printer

Ender 3 Pro

### Reproduction steps

Try to multiply an object several times

### Actual results

Nothing happend, but a label on the screen keepts always saying :
"Multiplying and Placing Objects"
[image]

### Expected results

Multiplyed objets appears on working area

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[icableguards-read-description-by-3dmx.zip](https://github.com/Ultimaker/Cura/files/7824976/icableguards-read-description-by-3dmx.zip)

MiguelEro, 2022-01-06:
I Update to 4.12.1 Version and Works fine.

closed 2022-01-06 by MiguelEro

Ghostkeeper, 2022-01-14:
Indeed, this was one of the reasons we put out version 4.12.1 (over 4.12.0).