Ultimaker/Cura #11213: Cura freezes while slicing

menoch0376, 2022-01-07:
### Application Version

4.12.1

### Platform

Windows 11

### Printer

Anycubic Chiron and Mega X Problem is on both

### Reproduction steps

Slicing files. I can slice files for a few days. Eventually, the program starts slowing down on slicing and eventually just freezes while slicing. I left a file slicing overnight, after slicing it in under 5 minutes and then making a small change. clicked slice. 15 minutes later the blue bar was 1/3 done. Woke up this morning and it was still in the same spot. I have closed out and reopened. removed and added the printer back, and even rebooted my computer. Nothing is working. This happened a few weeks ago, until an update resolved the issue. I've been waiting now for 11 hours to get this file sliced. That is a lot of print time lost. 

### Actual results

see above

### Expected results

The file should slice in a short time, it is only a 3 day print, and had sliced previously just fine until I changed support type and resliced

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I'm a relative newbie, but this seems to be a software issue if it resolves for a while and then returns with me doing nothing different than any other project. Searching help forums and many people are reporting the same issue with4.12.1. Please resolve.

fvrmr, 2022-01-07:
Hi @menoch0376 thank you for your report. 
Does it happen with every file? Could you share your project file (file - save project) and your log file? 
You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

menoch0376, 2022-01-07:
I have tried to save the file as project and upload and it only allows me
to save as 3MF and that isn’t a valid file type to upload. It’s only
happening with this larger file. Smaller stuff seems to be working fine

fvrmr, 2022-01-11:
You can share your 3mf on Github as a .zip. 
But I am wondering how large is your file?

menoch0376, 2022-01-11:
The problem eventually resolved. I had to uninstall and reinstall a few
times. Then reloaded mesh tools and it worked.

fvrmr, 2022-01-13:
Thanks for your reply. I will close this issue since it is solved!

closed 2022-01-13 by fvrmr

rockinjon49, 2023-04-29:
I too am having an issue where my 5.3.x as it happened on 5.3.0 and now on 5.3.1 where I can't slice anything.  Big or small the program stops responding after I click slice. Tried opening in admin  tried deleting the cura folder under the user profile

TaylorTheDeveloper, 2023-09-01:
I have also faced the exact same issue described here. Cura seems to have corrupted the local folder and I had to re-install to fix it.

rockinjon49, 2023-09-01:
i have uninstalled and reinstalled did not fix for me i was forced back to
4.1.0

Blip-Glok, 2023-12-22:
Hello, you really have this problem in the program, even in new versions, when a model with a large print volume the program freezes and loads the processor at 100 percent. It can be fixed? My Anycubic kobra MAX printer.
[image]
[image]