Ultimaker/Cura #11214: Not saving changes in extruder settings

JediRAMA, 2022-01-07:
### Application Version

arachne beta 2

### Platform

win10

### Printer

Kosher

### Reproduction steps

im changing number of cooliong fan from 0 to 3 and 5

### Actual results

It works well, but after restart cura changes reset to 0

### Expected results

in cura 4.13 that changes saving after restart cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

JediRAMA, 2022-01-10:
I added this printer from list in cura. Not custom printer

fvrmr, 2022-01-14:
I tried it again with your printer, but I still couldn't reproduce your issue. Maybe your log file could help. 
You can find your log file here: C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log

closed 2022-01-28 by no-response[bot]