Ultimaker/Cura #11216: ModuleNotFoundError sentry_sdk

dmitrygribenchuk, 2022-01-07:
### Application Version

4.13~202201071332~rev5557~pkg394~ubuntu20.04.1

### Platform

Linux Mint

### Printer

Creality Ender 3 v2

### Reproduction steps

Run cura

### Actual results

grid@nb000292:~$ cura -v
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp, stat
Traceback (most recent call last):
  File "/usr/bin/cura.real", line 24, in <module>
    from cura.CrashHandler import CrashHandler
  File "/usr/lib/python3/dist-packages/cura/CrashHandler.py", line 15, in <module>
    import sentry_sdk
ModuleNotFoundError: No module named 'sentry_sdk'

### Expected results

Starting cura

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log.gz](https://github.com/Ultimaker/Cura/files/7830903/cura.log.gz)

nallath, 2022-01-08:
It sounds like you didn't use the appimage. If so, whoever packaged Cura for your distro forgot to include a library.

dmitrygribenchuk, 2022-01-09:
Yes, I'm using the PPA repository. And it looks like dependency miss config.

fieldOfView, 2022-01-09:
Best report this to the maintainer of the ppa

fvrmr, 2022-01-10:
Closing this issue since it is not an issue with our version.

closed 2022-01-10 by fvrmr