Ultimaker/Cura #11217: Hole horizontal expasion problem

MLeal91, 2022-01-08:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Creality Ender-3 Pro

### Reproduction steps

I have set the horizontal expansion hole with a value of 0.1 my holes come out correct with this value. With this value active (0.1). My interior walls of the piece don't look right on the slicer

### Actual results

the walls of the piece were not correct

### Expected results

Changing value of the hole horizontal expansion option should only change hole diameters

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Link for Files https://files.fm/f/jcvw633kp

One other thing I would suggest but that is unrelated to the problem.
If it was possible to have a field in the speed options only for the hole features and the respective minimum and maximum values. I know this exists in cura but changing the value "Small Hole max size" influences the rest of the piece

This is my first report

smartavionics, 2022-01-08:
That's because the inside of your model is a big hole and therefore will be expanded/contracted when the hole horizontal expansion value is non-zero. Arguably, it would be useful to have additional setting(s) that would only expand/contract holes whose perimeter length is below/above some threshold value.

smartavionics, 2022-01-08:
Here's what I mean, I've added the (snappily named) setting "Hole Horizontal Expansion Max Path Len" to limit the expansion to holes whose perimeter length is less than that value. In the images below I've increased the hole expansion to 1mm to make the difference obvious...

Old behaviour by default when setting is 0

[image]

New behaviour when setting is 20

[image]

I've thought of doing this before but never got around to it but your issue has reminded me to do it, thanks!

GregValiant, 2022-01-08:
_"should only change hole diameters"_
This is from the Marlin description of G2/G3 but it applies to any calculation of a circular move:
_"Arc moves actually generate several short straight-line moves, the length of which are determined by the configuration option MM_PER_ARC_SEGMENT (default 1mm)."_
So unfortunately, there are no hole "diameters" - just polygons of varying sizes and shapes because any model file is made up of triangles.  PI has been calculated to 30trillion decimal places.  The line segments get shorter, and the measurement of the circumference and area get more accurate, but it's still a polygon and not a true circle.
@smartavionics good idea.  Do you have that in your version yet?  Allowing a user to enter a "diameter" rather than the circumference would be nice.  The "max path length" could be calculated in the background.

smartavionics, 2022-01-08:
OK, I've added a diameter setting to make it easy when the hole is circular. For non-circular holes, the user will just have to work out the perimeter length. This will be in my next release which will appear after 4.13 is released.

[image]

Happy now?

GregValiant, 2022-01-08:
Yessir, that'll do.  I get tired of my calculator falling off my lap.

MLeal91, 2022-01-08:
> OK, I've added a diameter setting to make it easy when the hole is circular. For non-circular holes, the user will just have to work out the perimeter length. This will be in my next release which will appear after 4.13 is released.
> 
> [image]
> 
> Happy now?

NIce thats it, i think the problem with small features holes disapear now.
but is possible to input a specific speed for machine do that small holes?
Thanks a lot

smartavionics, 2022-01-09:
> but is possible to input a specific speed for machine do that small holes?

Look at Small Hole Max Size and Small Feature Speed settings.

MLeal91, 2022-01-10:
> 

Sure but it is in % not i mm/s and I think it would be better on the speed tab

fvrmr, 2022-01-14:
Hi @MLeal91 thank you for your report. 
We are aware that this is indeed a bit confusing. So I have discussed with the team to add a similar setting like @smartavionics implemented. And I created a ticket on our backlog for this. 
Devs see CURA-8890
This would be a new feature so I am changing this to a feature request instead of a bug.

qwerty8224, 2022-01-17:
> ОК, я добавил настройку диаметра, чтобы было проще, когда отверстие круглое. Для некруглых отверстий пользователю просто нужно определить длину периметра. Это будет в моем следующем релизе, который появится после выхода 4.13.
> 
> [image]
> 
> Счастлив теперь?

you are great as always )