Ultimaker/Cura #11218: Multiplying objects does nothing.

JustinCase144, 2022-01-08:
### Application Version

4.12.1

### Platform

Garuda Linux with latest kernel-zen

### Printer

Anycubic Mega Zero 2

### Reproduction steps

I had a small part stl open and I wanted to multiply it so I could print three at a time. 
I hit ctrl+M and chose two more and hit ok. 
[image]

### Actual results

Cura just opens a box to say "Multiplying and placing objects", with a loading bar beneath, but does absolutely nothing else. 
I can do this many times and it just opens a new box every time.
[image]
 

### Expected results

I was expecting for two more parts to appear. 

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7832512/cura.log)
Here is the logfile. I see some errors related to the issue at hand, but I cannot understand it.

dinuxdotde, 2022-01-08:
Same on my System, Arch Linux, Cura 4.12.1

protist, 2022-01-08:
I also have this issue. I'm also on Arch Linux with Cura 4.12.1. I get the following error when running from CLI.

```
2022-01-09 00:39:16,087 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]: Exception: Job <cura.MultiplyObjectsJob.MultiplyObjectsJob object at 0x7fa0416ebd00> caused an exception on worker JobQueueWorker [3]
2022-01-09 00:39:16,087 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]: Traceback (most recent call last):
2022-01-09 00:39:16,087 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 133, in run
2022-01-09 00:39:16,087 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     job.run()
2022-01-09 00:39:16,087 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/MultiplyObjectsJob.py", line 79, in run
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     group_operation, not_fit_count = createGroupOperationForArrange(nodes,
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 119, in createGroupOperationForArrange
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     found_solution_for_all, node_items = findNodePlacement(nodes_to_arrange, build_volume, fixed_nodes, factor)
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 94, in findNodePlacement
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     converted_points.append(Point(point[0] * factor, point[1] * factor))
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]: TypeError: Point(): arguments did not match any overloaded call:
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   overload 1: argument 1 has unexpected type 'numpy.float64'
2022-01-09 00:39:16,088 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   overload 2: argument 1 has unexpected type 'numpy.float64'
```

nallath, 2022-01-08:
Did you use the appimage?

protist, 2022-01-09:
Previously I built from source, which failed. I just tested the appimage, and this works as expected.

JustinCase144, 2022-01-09:
> Did you use the appimage?

No. Used the arch repo package.

fvrmr, 2022-01-10:
Closing this issue since it is not an issue with our version. 
Please reach out to the maintainer of this package.

closed 2022-01-10 by fvrmr

protist, 2022-01-10:
The Arch issue is [here](https://bugs.archlinux.org/task/73327).

Ghostkeeper, 2022-01-17:
A fix for this is also planned in our own source code here: https://github.com/Ultimaker/Cura/pull/11196