Ultimaker/Cura #11219: Temperature at first layer with multiple parts

EddyCoubard, 2022-01-08:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Dagoma Disco Ultimate bi-couleur

### Reproduction steps

When you have multiple parts, and you use the function (name in French) "Séquence d'impression : Un à la fois", the function "Température d'impression couche initiale" only works for the first part, not for the following parts.

### Actual results

Only the first part has got its first layer with the right temperature.

### Expected results

All the parts should have their first layer with the right temperature.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-01-08:
Good one.  "Build Plate Temperature Initial Layer" as well.

fvrmr, 2022-01-14:
Hi @EddyCoubard thank you for your report. 
I have discussed this with the team and I created a ticket on our backlog for this. 
Devs see CURA-8889

tlhintoq, 2022-11-13:
Bug reported in version 4.
Bug still present in version 5

tlhintoq, 2022-11-13:
https://www.facebook.com/groups/1959877270949115/posts/3345914442345384

Does this sound like a bug to anyone else?
Initial layer temp: 215
Print temp: 200
{Because you need a higher temp to get good adhesion to bed}
Multiple items printing in "one at a time" mode.
Initial layer temp of 215 *should be* at the initial layer of each model, right?
Looking at gCode I'm seeing its only at the very first layer 0 of the first object.  All the other objects are trying to stick to the glass bed at 200 instead of the initial layer temp of 215

tlhintoq, 2022-11-13:
Here's the job I'm running.  5 coasters one at a time.
Initial layer temp 215
Regular layer temp 200
[image]

And here's the searched code.  Only temp changes take place at startup, first object, and end of job.
[5x_HoneycombCoaster_PETG_9hrs.gcode.txt](https://github.com/Ultimaker/Cura/files/9998307/5x_HoneycombCoaster_PETG_9hrs.gcode.txt)

[image]

Full gcode attached as well (just take off the .txt extension)

GregValiant, 2022-11-13:
I see that fvmr noted that this is on the backlog as CUR-8889.  I have no idea where this is in the process or if it has been assigned.
What I do have is a possible workaround.
2 instances of Search and Replace.
Instance 1:
Search = ;LAYER:0\n
Replace = ;LAYER:0\nM109 S215\n
Use Regular Expressions = checked

Instance 2:
Search = ;LAYER:1\n
Replace = ;LAYER:1\nM104 S200\n
Use Regular Expressions = checked

The "\n" is a newline character and they are required as shown.

Cura will replace the ";LAYER:0" line with two lines:
;LAYER:0
M109 S215

So at every Layer:0 the printer will stop and wait for the hot end to get to 215 and then resume printing.
At every layer 1 the temperature will go back to 200 but the M104 will not hold up the print progress.

tlhintoq, 2022-11-13:
Trial permutation: No copies of objects within Cura.
Brought the same STL in from windows 5 individual times.
Same result: No changes in the gCode.
[image]

tlhintoq, 2022-11-13:
Also, 5 unique files of different objects do the same thing

GregValiant, 2022-11-14:
Did you follow my workaround?
Go to "Extensions / Post Process / Modify Gcode" and then "Add a Script" and add two Search and Replace scripts.  Follow the instructions in the previous post.
You will end up with this dialog.  The red circle next to the slice button indicates when you have post-processors active.  I can see you don't have any.  (My images says "4" because I always have two active at all times.)
[image]

tlhintoq, 2022-11-14:
@GregValiant 

Great use of the post-process gCode.  Gotta love RegEx!

GregValiant, 2022-11-14:
One of the reasons I really like Cura is that it is so highly customizable.  Fooling around here as the "greeter" I have learned a lot about the program and what it is capable of.
It isn't real intuitive, the "Help" isn't very good (writing and updating help articles in about 40 languages would be an awful chore), and there is certainly a learning curve but at the end of the day I have found that the time I invest is worth it.

MariMakes, 2023-08-29:
Quick update from our side 👋

One of our developers took a look, and it seems to be resolved for a large number of printers in Cura 5.4.
Could you test if that's the case for you as well? 
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.4.0

If you still run into the issue, please let us know what printer you are using and we can doublecheck if the fix we are working on for 5.5 is also applicable to your printer. 👍

tlhintoq, 2023-08-30:
> Quick update from our side 👋
>
> One of our developers took a look, and it seems to be resolved for a large number of printers in Cura 5.4. Could you test if that's the case for you as well? You can [...]

First let me say "thank you" for confirming and acknowledging this a genuine issue that you were able to find in code and affect a fix.  Just knowing I'm not imagining it does wonders.

As for machines:
I see this on an Elegoo Neptune3max - and a handful of machines that derive from that.  
IE: I make an Ender3pro that derives from Neptune3max with changed bed size so that all my print profiles move between the machines.
But I also see this in a Hictop D3Hero, which should derive from an Tenlog since the Hictop is just a rebranded Tenlog.

MariMakes, 2023-10-11:
Good news 😄 

This is resolved in the 5.5 Beta release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.5.0-beta.1)
I'll close this issue since it's resolved. 

Thanks again, and please let us know if you run into any other issues 💪

closed 2023-10-11 by MariMakes

tlhintoq, 2023-10-11:
@MariMakes 
Just read through the 5.5b1 write up.  I know I'm quick to snap but I try to be equally quick to praise: This looks to be one of the biggest collection of fixes & improvements that will genuinely affect users, fix issues and help make truly better printed parts, that I've seen in a long time.  I know all work is important but so many things here that are less about 'under the hood' and will directly make life better for us like grid layout, top surface tuning, small top, {...}
I have to say I'm *genuinely* excited to try this.
Thank you!