Ultimaker/Cura #11220: Incorrect bridging on single layer

ebenolson, 2022-01-08:
### Application Version

4.12.1

### Platform

Ubuntu 18.04

### Printer

Ender 3 Pro

### Reproduction steps

Loaded and sliced attached STL. 
[y slider block with motor.zip](https://github.com/Ultimaker/Cura/files/7833536/y.slider.block.with.motor.zip

### Actual results

There is an unexpected triagular bridged area on a single layer:

[image]
)

### Expected results

Should not have that triangular bridge.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

Project file:
[CE3_y slider block with motor.zip](https://github.com/Ultimaker/Cura/files/7833540/CE3_y.slider.block.with.motor.zip)

GregValiant, 2022-01-08:
This has come up before on the Ultimaker Forum.  It was unresolved as only the poster was having the problem, but for him it was consistent.  The thread is [HERE](https://community.ultimaker.com/topic/36536-cura-generates-geometry-not-present-in-the-original-file/).  This is a box and all 5 internal triangles are wrong.
[image]

ebenolson, 2022-01-08:
Thanks for the link. Like that poster, I'm exporting from solidworks and formware mesh repair tool fixed some things that seem to have resolved it. Guess I just can't trust the STL export.

closed 2022-01-08 by ebenolson

smartavionics, 2022-01-09:
I played around with this a bit and discovered the following...

1 - changing the inner line width to 0.61 removed the badness.

2- I think it's related to simplification of the inner wall polygons because the following patch also appears to cure the issue.

```
diff --git a/src/WallsComputation.cpp b/src/WallsComputation.cpp
index d6c526fd4..6221e10e0 100644
--- a/src/WallsComputation.cpp
+++ b/src/WallsComputation.cpp
@@ -114,7 +114,10 @@ void WallsComputation::generateInsets(SliceLayerPart* part)
         const Settings& resolution_settings = wall_extruder_nr < 0 ? settings : Application::getInstance().current_slice->scene.extruders[wall_extruder_nr].settings;
         const coord_t maximum_resolution = resolution_settings.get<coord_t>("meshfix_maximum_resolution");
         const coord_t maximum_deviation = resolution_settings.get<coord_t>("meshfix_maximum_deviation");
-        part->insets[i].simplify(maximum_resolution, maximum_deviation);
+        if (i == 0)
+        {
+            part->insets[i].simplify(maximum_resolution, maximum_deviation);
+        }
         part->insets[i].removeDegenerateVerts();
         if (i == 0)
         {
```
i.e., only simplify the outer wall polygon and not the inners.

Hope this helps.

smartavionics, 2022-01-09:
A better patch...
```
diff --git a/src/WallsComputation.cpp b/src/WallsComputation.cpp
index d6c526fd4..37cad8802 100644
--- a/src/WallsComputation.cpp
+++ b/src/WallsComputation.cpp
@@ -109,12 +109,15 @@ void WallsComputation::generateInsets(SliceLayerPart* part)
             }
         }
 
-        //Finally optimize all the polygons. Every point removed saves time in the long run.
-        int wall_extruder_nr = settings.get<int>(i == 0 ? "wall_0_extruder_nr" : "wall_x_extruder_nr");
-        const Settings& resolution_settings = wall_extruder_nr < 0 ? settings : Application::getInstance().current_slice->scene.extruders[wall_extruder_nr].settings;
-        const coord_t maximum_resolution = resolution_settings.get<coord_t>("meshfix_maximum_resolution");
-        const coord_t maximum_deviation = resolution_settings.get<coord_t>("meshfix_maximum_deviation");
-        part->insets[i].simplify(maximum_resolution, maximum_deviation);
+        if (i == 0)
+        {
+            //Finally optimize all the polygons. Every point removed saves time in the long run.
+            int wall_extruder_nr = settings.get<int>("wall_0_extruder_nr");
+            const Settings& resolution_settings = wall_extruder_nr < 0 ? settings : Application::getInstance().current_slice->scene.extruders[wall_extruder_nr].settings;
+            const coord_t maximum_resolution = resolution_settings.get<coord_t>("meshfix_maximum_resolution");
+            const coord_t maximum_deviation = resolution_settings.get<coord_t>("meshfix_maximum_deviation");
+            part->insets[0].simplify(maximum_resolution, maximum_deviation);
+        }
         part->insets[i].removeDegenerateVerts();
         if (i == 0)
         {

```

Ghostkeeper, 2022-01-17:
This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/10369.