Ultimaker/Cura #11221: [BUG] (?) Cura 4.11.x and above: Abort prints in Cura / Marlin when LinearAdvance is activated

SpRiNgToIfEl, 2022-01-08:
### Application Version

4.11.x and above

### Platform

Win10Pro (newest Updates installed)

### Printer

Ender 5 Plus with BTT SKR PRO 1.2 / BTT TFT50v3 / Marlin 2.9.0.1 and above

### Reproduction steps

1. Installed Cura 4.11.x / 4.12.x / 4.13.0b
2. Installed [LinearAdvance Plugin](https://github.com/fieldOfView/Cura-LinearAdvanceSettingPlugin) (latest version)
2. Sliced the Thingiverse Calibration Cube [(Link Thingiverse)](https://www.thingiverse.com/thing:1278865) in Cura 4.11.x and above 
3. Start Print: Ender 5 Plus with BTT SKR PRO 1.2 / BTT TFT50v3 / Marlin 2.9.0.1 and above.

### Actual results

- In Marlin LinearAdvance ist activated (see Files).
- In Cura the PlugIn "LinearAdvance" is installed

Test 1:
Sliced in Cura 4.11.x and above **without** Linear Advance (0.0): Print is sucsesfull
Sliced in Cura 4.11.x and above **with** Linear Advance (>0.0): Print fails after 4-5 layers (Position of fail: random position in outer wall,  inner wall or infill)

### Expected results

*

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Configuration_2.9.0.1.zip](https://github.com/Ultimaker/Cura/files/7833886/Configuration_2.9.0.1.zip)
[Configuration_2.9.0.3.zip](https://github.com/Ultimaker/Cura/files/7833887/Configuration_2.9.0.3.zip)
[image]
[image]

Asterchades, 2022-01-09:
What stepper driver you using on your extruder? If it's a TMC2208, this is expected - there's an internal communication issue in Marlin when LA is enabled which effectively results in the extruder driver shutting down after a short period. I'm taken to believe there are work-arounds involving the use of adjusting the pulse length and/or enabling square wave pulsing, but these are band-aid fixes at best.

You'll either have to sort out one of those fixes (which may or may not work), switch the extruder driver for pretty much anything else (TMC2209s are unaffected), or disable LA (setting it to 0.0 effectively disables it without new firmware).

SpRiNgToIfEl, 2022-01-09:
I am using TMC 2208v3 UART.

After some searching I found a "solution" to use LA on Marlin v2.9.0.3 (nightly):
- **activate** SQUARE_WAVE_STEPPING (beta)
- **deactivate** extruder STEALTHCHOP (extruder runs now with spreadcycle)

Now the e-stepper motor hums / beeps a bit (not annoying), but I can now print with LA.
If necessary, the TMC 2208 will changed in the future.

The background noise (extruder) should probably be reduced with higher steps on the extruder.

fvrmr, 2022-01-14:
Closing this issue since this is not related to Cura.

closed 2022-01-14 by fvrmr