Ultimaker/Cura #11222: Slow and erratic behaviour  of CURA interface while moving model

grandey-HA, 2022-01-08:
### Application Version

4.12.1

### Platform

Mac OS Monterey 12.0.1 (M1) 

### Printer

Artillery Sidewinder X1

### Reproduction steps

Import a STL/3MF in the cura instance. 

I tried to mode it...VERY SLOW...

I sliced it as usual (normal spped).
Time preview of the print is very very very slow....

### Actual results

cura is behaving normally except for moving the model or moving in the preview

### Expected results

the model should move easiliy as it used to do...

I believe the new Q engine is at fault, as all characters now appear ok (cooridantes, menu etc).

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://user-images.githubusercontent.com/73668739/148661369-cf0624d0-06cc-47c9-a22b-d4d7f481d7e7.mp4

this is a file showing the behaviour.

fvrmr, 2022-01-14:
Hi @grandey-HA thank you for your report. 
We don't fully support M1 yet. In the 5.0 release we will update our Qt libraries, which hopefully solves this issue.

Ghostkeeper, 2022-01-17:
Should be noted that the newest Qt library also doesn't support Monterey yet (but does support the M1).