Ultimaker/Cura #11223: Import Scale Issue

LtLasagna, 2022-01-08:
### Application Version

14.12.1

### Platform

Windows 10 / Ryzen 9 3900x

### Printer

FLsun SR

### Reproduction steps

1. Create something in Blender
2. Export as STL with attached scale
3. Import into 3d Builder to fix mesh
4. Save as .3mf
5. Open in Cura

### Actual results

If enabled, Cura will auto-scale up by a factor of 10000000x.
If automatic scaling is turned off the scale of the object will show as 0% instead of 100%

### Expected results

When opening files from 3d Builder the scale should be imported correctly as well.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[FSR_Barley Bulb.zip](https://github.com/Ultimaker/Cura/files/7834202/FSR_Barley.Bulb.zip)
I had to change the .3mf to a .zip in order to upload it, I'm sure you know that has no effect on the model.

[cura.log](https://github.com/Ultimaker/Cura/files/7834215/cura.log)

fvrmr, 2022-01-14:
Hi @LtLasagna thank you for your report.
I will bring this up for discussion. Keep you posted.

LtLasagna, 2022-01-14:
Just a note:
The scale in blender is set to millimeters.

Ghostkeeper, 2022-01-17:
The coordinates in this file are huge. This is the first one.

    <vertex x="525150" y="603696" z="3876460" />

That indicates an X and Y position about half a kilometer away from the front-left of the build plate. And 3.8km into the sky.

So I think the issue is more with 3D Builder, or whatever created that 3MF file, than with Cura.

fvrmr, 2022-01-18:
We have discussed it with the team and decided to defer this, because indeed the changes are that this is more related to 3D builder.
I hope you understand!

LtLasagna, 2023-05-26:
I haven't had this issue, but that's because I have changed my workflow since then.

closed 2023-05-26 by LtLasagna