Ultimaker/Cura #11225: Slicing model fails every time regardless of checking for items suggested

EgrStallion, 2022-01-09:
### Application Version

4.12.1

### Platform

Windows 10 64-Bit

### Printer

Creality Ender 3 Pro

### Reproduction steps

Set setting to 100% infill, low quality
Tried to slice part

### Actual results

received error message "unable to slice"

"Please review settings and check if your models:
-Fit within the build volume
-Are assigned to an enabled extruder
-Are not all set as modifier meshes"

### Expected results

Slicing should have occurred.  I did not play with extruder settings at all, and my build volume was well under 220 x 220mm.  (Not sure what a modifier mesh is, though.)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CE3PRO_Rectangular Pot.zip](https://github.com/Ultimaker/Cura/files/7834462/CE3PRO_Rectangular.Pot.zip)
[image]

GregValiant, 2022-01-09:
The gray border is reserved for the skirt/brim and you can see your model intrudes into that area.  You have a skirt Line Count of 3 and a Skirt Distance distance of 10mm and that adds to the footprint size of the model on the build plate.  So the 215 model + (1.2 + 10) on each end = 237.4.
Change the Skirt Line Count to 1 and the Skirt Distance to 2mm and the footprint becomes 219.4 and you should be good.

EgrStallion, 2022-01-10:
Thank you, Greg!  I obviously am still becoming familiar with the
software.  I was finally able to get it to slice due to your advice.  I
noted how to go through all the other advanced settings now as well
(albeit, will need to come up to speed on a few of them).

fvrmr, 2022-01-10:
Closing this issue since this is not a bug but related to settings. 
Feel free to reach out if you have more questions!

closed 2022-01-10 by fvrmr