Ultimaker/Cura #11226: cannot open "help/show configuration folder"

amills1982, 2022-01-09:
### Application Version

4.12.1

### Platform

ubuntu 20

### Printer

ender 3

### Reproduction steps

click help
click " show configuration folder"
nothing happens
trying to install "createJPEGthumbnail.py" script

### Actual results

nothing happens
trying to install "createJPEGthumbnail.py" script

### Expected results

config folder should open

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

works fine on 4.4.1

fvrmr, 2022-01-10:
Hi @amills1982 thank you for your report.
Are you using the AppImage?

amills1982, 2022-01-10:
It’s the snap installer, I’m going to pause this (for me) for the moment as I’m going to install mint tonight and try cura on that os

 Snap

fvrmr, 2022-01-11:
Thank you for your reply. I will close this issue since it is not an issue with our AppImage. Please report your issue to the maintainer of this package.

closed 2022-01-11 by fvrmr

amills1982, 2022-01-11:
Installed mint last night. Cura config folder is accessible. Ty for your responses

Closed #11226<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2FUltimaker%2FCura%2Fissues%2F11226&data=04%7C01%7C%7C6a00992e57334e29283508d9d4eddb34%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637774940323226521%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=h34r9tbcQZw92yZMxEv6wPCG0iEHOfk5XUcD47QCh08%3D&reserved=0>.