Ultimaker/Cura #11227: Cannot resize add printer dialog, making verifying gcode impossible.

thantik, 2022-01-09:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Any.

### Reproduction steps

1. Click Add Printer from the Printers drop down on the left.
2. Add any printer.
3. Dialog box comes up with machine geometry, nozzles, etc

### Actual results

Window is not resizable because of incorrect window designation and therefore only a single line of gcode is viewable at a time.  Paired with the terrible non-standard text control box, you cannot scroll to a specific line and thus have to _finish_ adding the printer, and then re-open the manage printers dialog box, and click "Machine settings" to have the dialog displayed again.

Upon the _SECOND TIME_ opening this dialog box, the window is resizable so that you can expand the two text boxes and verify the correct start/end gcode.

### Expected results

Add Printer dialog box should be resizable and/or by default - a larger size, so that the text boxes are not so small that you cannot functionally use them.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

https://i.imgur.com/dq2J3z2.png - This dialog is unable to be resized.

https://i.imgur.com/7qKmTOy.png - However this dialog is actually able to be used, because not only is it resized, but it has text controls that are larger than 2 lines tall.

fvrmr, 2022-01-14:
Hi @thantik thank you for your report.
I can reproduce your issue, and it turns out only the add machine window is not resizable. 
I will bring this up for discussion, keep you posted.

opok, 2022-01-20:
Same behavior on Linux Manjaro, Cura 4.13.0 installed from AppImage
[image]

GregValiant, 2024-11-16:
This bug appears to have been fixed in later versions of Cura (5.8.0 and up).  Can this be closed?

thantik, 2024-11-16:
Can verify it's been fixed on 5.6.0 as well.

closed 2024-11-16 by thantik