Ultimaker/Cura #11228: supports out of scale and orientation

sir-ustad1, 2022-01-09:
### Application Version

4.11.0

### Platform

windows 10

### Printer

creality ender 3 pro and v2

### Reproduction steps

imported model scaled model first couple edits were fine. after a couple times went to add more supports and the supports are scaled down to the model scale not the size set in supports window, also the supports orientation are not to the print bed they are towards where the bottom of the model should be not the bed

### Actual results

[image]
[image]

### Expected results

support should be at support menu size and should orient to the build plate

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[bug1.zip](https://github.com/Ultimaker/Cura/files/7835467/bug1.zip)

fvrmr, 2022-01-10:
Hi @sir-ustad1 thank you for your report.
Are you using the Cylindric Custom Support plugin or Custom support plugin? I think it might be related to these plugins and not to Cura.

sir-ustad1, 2022-01-11:
I am using both

fvrmr, 2022-01-14:
So does this happen when you want to create custom support?

Ghostkeeper, 2022-01-18:
Do I understand here that the issue is that the models got rotated/rescaled when saving and re-loading from 3MF?

Or is it just that you tried to rotate it and it didn't rotate the way you liked or something?

sir-ustad1, 2022-01-18:
No, it doesn't. I'm using stl,  I scale (up or down) add
supports(cylindrical or custom),, and the support is either what the
unsealed model size(too big or too small) and off in a direction other than
to the build plate,, its not all the time but it is happening about 15% of
the time

sir-ustad1, 2022-01-18:
Oh I know it is,, doesn't make alot of sense , ,very frustrating

fvrmr, 2022-01-28:
We have discussed this with the team and decided to defer this. It will not get a high priority to fix for us, since there are more important issue we want to fix. I hope you understand.

closed 2023-06-10 by github-actions[bot]

sir-ustad1, 2023-06-10:
It's an occasional problem when editing older saved projects