Ultimaker/Cura #11229: Cura 4.13.0-BETA recommending me to update to 4.13.0-BETA

hapklaar, 2022-01-09:
### Application Version

4.13.0

### Platform

WIndows 10

### Printer

Biqu B1

### Reproduction steps

Start Cura 4.13

### Actual results

Observe popup below" Ultimaker Cura 4.13.0-BETA is available!"

[image]

[image]

### Expected results

Don't recommend upgrading to version 4.13.0-BETA when that version is already being used.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7835670/cura.log)

fvrmr, 2022-01-10:
Hi @hapklaar thank you for your report. 
We discovered this as well and have ticket on our backlog for this. 
Devs see CURA-8856

Jarren141, 2022-04-22:
I'm also seeing this on 5.0.0-beta+1

fvrmr, 2022-05-11:
Closing this issue, since it is fixed for next beta releases!

closed 2022-05-11 by fvrmr