Ultimaker/Cura #11230: Automated purge tower location

65643243, 2022-01-09:
### Is your feature request related to a problem?

i would like to propose a feature that automatically places the purge tower close to my print object. 

normally the purge tower is placed in the back right corner of the construction room. this way the purge tower is as close as possible to the point where the nozzle switch takes place. 
However, it makes more sense to place the purge tower close to the print object so that the travel time between the purge tower and the object is as short as possible to prevent oozing. 

### Describe the solution you'd like

at the moment you have to manually set the x and y coordinates of the purge tower to move it. this is quite cumbersome every time. 
a function to automatically place the purge tower close to the object, selectable from back, right, front, left would save time and make it easier to set up a print. 
here i would suggest a drop down menu which places the purge tower in the usual position or in different positions directly near the print object with 1 or 2 mm distance. 

### Describe alternatives you've considered

at the moment, the purge tower can only be moved using the x and y coordinates, which means slowly approaching the best position. 

### Affected users and/or printers

i think most users place the purge tower individually. especially for small parts, a particularly long empty run towards the object makes little sense, as the oozing interferes here. a longer travel move to the purge tower after the nozzle switch, on the other hand, does not interfere, as the nozzle strips off any possible excess material here. 

### Additional information & file uploads

how it makes sense to position the purge tower:
[image]

how the purge tower is currently positioned:
[image]

fvrmr, 2022-01-10:
Hi @65643243 thank you for your feature request.
We have a similar ticket on our backlog related to the position of the prime tower.
I will link this request to the ticket.
Devs see CURA-8382

ThePaleBlueBot, 2023-04-25:
Hi @fvrmr, 
I would like to know if this request is still active. 
I want to create a new request concerning dual extrusion and prime tower, and I don't want to ask that same question twice.
Thanks for your help.

ThePaleBlueBot, 2023-05-09:
Hello @GregValiant, Hello @MariMakes,

I'm mentioning you two because it seems you know everything☺️.
I'm sorry if it's not how to do it. I haven't found a way to make this subject appear on the front page again.

I'm interested by the idea of @65643243
Can you tell me if this idea will be added to Cura?

It would be a fantastic addition for people that use multi-extruder printers.

Thank you.

MariMakes, 2023-05-10:
Hey @ThePaleBlueBot,

We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs.

I double-checked our backlog and it seems to have been closed to be able to focus on the most requested and needed features and bug fixes from our community and from Ultimaker. Unfortunately, this issue was never updated to reflect this correctly. 

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.
I'll be closing this issue, I hope you understand.

ThePaleBlueBot, 2023-05-11:
@GregValiant, I agree. Being able to position it manually would be enough.