Ultimaker/Cura #11233: Fill pattern

hanheik, 2022-01-10:
### Is your feature request related to a problem?

no problem

### Describe the solution you'd like

[image]
Hi, i think a pattern like they use for 3D-printing houses would be helpful for cylindrical objects. Just a line connecting the inner and and outer wall.
ys
HHK

### Describe alternatives you've considered

no alternatives

### Affected users and/or printers

all users

fvrmr, 2022-01-13:
Hi @hanheik thank you for your feature request.
You can do this with the zig zag infill pattern.
[image]
Select zigzag as infill pattern and change the infill direction with one value so all the lines will go in the same direction.
Let me know if that works for you!

fieldOfView, 2022-01-14:
Though that somewhat works in the top and right of that model, it does not work in the left.

fieldOfView, 2022-01-14:
The requested infill would entail finding the centerline (medial axis or topological skeleton) of the area needing infill, and having a sine or sawtooth modulate over that path by the local feature size.

Ghostkeeper, 2022-01-18:
While it's easy to think of shapes where this infill pattern would be simple, it's hard to think of a way to generate this infill pattern for all shapes. Especially if the model contains splits and a big difference between inner and outer radii this is problematic.

Essentially, this suggestion suffers from the same design problems as https://github.com/Ultimaker/Cura/issues/8811.

closed 2022-01-27 by no-response[bot]