Ultimaker/Cura #11234: Slice incomplete

DeCaff24, 2022-01-10:
### Application Version

4.12.1 and 4.8.0

### Platform

Windows 10

### Printer

Malyan M200

### Reproduction steps

1: I made a test cube 20x20x20mm with a few fillets. I added debossed "Text" to one side and embossed "Text" to the (top)side.
2: saved it to stl and alligned it as described above in Cura 4.12.1 and 4.8.0. Sliced it and printed it. 
3: Printer started on making the embossed first T but stopped there.
4: I checked the slicer and saw that Cura didn't completely finish the Text on top.

[image]
stl file

[image]
prepare screen Cura

[image]

Slicer Preview. see that it diDn't include the embossed text on the top

### Actual results

Printed object was missing the embossed text

[image]

### Expected results

The slicer should have included the embossed text and it should have been on the printed object

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[MM_Cube 20x20.zip](https://github.com/Ultimaker/Cura/files/7840055/MM_Cube.20x20.zip)

fvrmr, 2022-01-13:
Hi @DeCaff24 thank you for your report.
This is probably happening because your text is too thin. Slicing your model in our Arachne release, it will show the text perfectly. Because with Arachne uses variable line width. 
[image]

Read here more about the Arachne release: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2

I will close this issue since we planned to implement the Arachne Engine in Cura 5.0
I hope you understand!

closed 2022-01-13 by fvrmr

Ghostkeeper, 2022-01-18:
For the stable release, Print Thin Walls may also help (but tends to produce worse results than Arachne for such things).