Ultimaker/Cura #11238: SD card is not written proper G code from Cura

Corallus-Caninus, 2022-01-10:
### Application Version

4.9.0

### Platform

Windows 10

### Printer

Anet

### Reproduction steps

wrote to micro SD card and ejected but printer receives bad G code. and motors are started on print start instead of setting the temperature and waiting.

### Actual results

printer started making noises as if it received incorrectly formatted G code.

### Expected results

The print should start correctly.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

If I rewrite to the sd card exactly twice, ejecting each time the print runs correctly.

GregValiant, 2022-01-11:
I have gotten this a few times.  If there are bad sectors on the memory card then parts of the file can be written to those bad sectors and are not readable by the printer and so they are simply ignored.  The printer then starts at whatever the first line it understands is.  In your case it appears to be down lower into the model gcode.
Did you open the Gcode file (in a text editor) that is on the SD card?  Does it look like a text file or does it look like hexadecimal gibberish?
Have you formatted that SD card recently?  That can often eliminate the problem.  Using a different memory card can cure the problem as well (until it happens with the new card).

Corallus-Caninus, 2022-01-24:
formatting seemed to fix this issue, I understand this can be a sign of failing SD cards but pasted here since writing and ejecting twice to the SD card seemed to fix the problem.

closed 2022-01-24 by Corallus-Caninus