Ultimaker/Cura #11239: Tree Supports not covering open areas

nightmage80, 2022-01-10:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

AnyCubic Chiron

### Reproduction steps

Use Tree Supports
Branch Angle 45
Support Overhang 55

Make sure have a hand or a limb that is out at about a 45 angle or less

### Actual results

Limb is not supported

### Expected results

Limbs should be supported

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[cura0110.log](https://github.com/Ultimaker/Cura/files/7841850/cura0110.log)
[image]

fvrmr, 2022-01-13:
Hi @nightmage80 thank you for your report.
Could you share your project file (file - save project) and share it as a .zip here on Github.

nightmage80, 2022-01-13:
Sorry about closing, it would not let me leave a comment without closing. May have done something wrong. Let me know if you need the GCode as well. When I add the zip, it says "is not include in the list" is there another way to send it to you?

fvrmr, 2022-01-13:
You can also use wetransfer or a google drive/dropbox link!

nightmage80, 2022-01-13:
https://drive.google.com/drive/folders/14TPMusDzD9KRru7VwfrErPKM-k7xbkSC?usp=sharing

Let me know if that works.

Ghostkeeper, 2022-01-18:
It works. Though I'm getting a somewhat different result, it still shows support missing a few parts.
[image]

(Pretty model by the way, although it would probably print better if it weren't hollow.)

When Support Placement is set to Everywhere (instead of Build Plate Only) the issue is gone. I think the problem is that it starts too close to the model, and being inside of the model it marks that support as not being able to reach the build plate.

Support Placement to Everywhere is not nice for this model since it is hollow. I would print this with the Remove All Holes option enabled, so that the model is no longer hollow (do not forget to turn that setting off afterwards though). It saves you a lot of wall printing in this case (like 30% reduced printing time in total), and allows you to place support everywhere.

The issue itself is still a valid issue that needs to be looked at though.

fvrmr, 2022-01-21:
We have discussed this with the team and linked this issue to a ticket on our backlog.
Devs see CURA-8840

DerGenaue, 2023-07-04:
This issue should be solved in the new Cura 5.4 release