Ultimaker/Cura #11240: Circles are getting printed very slow - sometimes

tispokes, 2022-01-10:
### Application Version

4.12.0

### Platform

Windows 11

### Printer

TT Sapphire Plus SP-5

### Reproduction steps

Sliced a STL file which is round in 80% of the layers

### Actual results

Most round/circle layers are printed very slow at maximum 10 mm/s. 
[image]

### Expected results

Some circles have faster printing speed, expected all to be printed faster as set up.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Maskenhalter_Helm_v2.zip](https://github.com/Ultimaker/Cura/files/7842100/Maskenhalter_Helm_v2.zip)

fieldOfView, 2022-01-10:
Have you seen the "minimum layer time" setting? Here's a good explanation:
https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/cooling/cool_min_layer_time.md

tispokes, 2022-01-10:
It's not a bug, it's a feature :-D Thanks @fieldOfView and also for your great addons.
Closed

closed 2022-01-10 by tispokes