Ultimaker/Cura #11241: File>Open or the Open file icon opens explorer dialog that crashes Cura

kylane, 2022-01-11:
### Application Version

4.12.1 & 4.13.0-BETA

### Platform

Win10

### Printer

Creality Ender 3

### Reproduction steps

Open Cura, click on the open file icon or go to menu file > open - opening the explorer dialog to select a file crashes the program.

### Actual results

Crash to desktop.

### Expected results

Be able to select a file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Screen recording: https://www.youtube.com/watch?v=OH3BzNhPM1

4.13.0-BETA - [cura.log](https://github.com/Ultimaker/Cura/files/7843398/cura.log)

4.12.1 - [cura.log](https://github.com/Ultimaker/Cura/files/7843402/cura.log)

nallath, 2022-01-11:
See https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file for a solution.

closed 2022-01-11 by nallath

kylane, 2022-01-11:
> See https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file for a solution.

Thanks Jaime!