Ultimaker/Cura #11244: Garbled G code

Pathoman, 2022-01-11:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

ANET A8

### Reproduction steps

1. I sliced an object, and stored the g-code on disk.
2. When I attempted to view the code in Cura, it
[image]
[image]
 

### Actual results

Garbled display.

### Expected results

I should be able to previev my print process

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Hello,

I was Happy with the previous version of Cura, and only experienced this problem after updating to 4.12.
[image]
[image]
1

fieldOfView, 2022-01-11:
Do you perhaps have the ArcWelder plugin installed and enabled? If so, Cura can not display the arc movements that are now in your gcode.

Pathoman, 2022-01-11:
Hello,

 

Thank you, that was the problem.

 

Thank you for the help.

 

Regards

 

Patrick Homan

078 712 7534

fvrmr, 2022-01-13:
Closing this issue, since it is solved!

closed 2022-01-13 by fvrmr