Ultimaker/Cura #11247: Model missing from Slice

cfisher273, 2022-01-11:
### Application Version

4.13.0

### Platform

Mac OS

### Printer

Ender 5 Pro

### Reproduction steps

This happened immediately after installing the app. 

I loaded this stl file from Blender. I have not had any previous issues doing this in several years. I have also tried other files from blender and web and they all work fine. It has something to do with this particular mesh. 

Load the model. Lay it flat and slice. 

Time: 0 min, 0g, 0m and $0.00 of material. 
The preview only shows a small brim in the center of the model on the platter. There is nothing else printed. 

### Actual results

Nothing. There is nothing sliced to print. 
If I lay the model on the side, it will print. However, something is not accepted by Cura in this model and it cannot calculate the necessary gcode. 

### Expected results

It should build/slice and provide the necessary g-code. 

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Booster.3mf.zip](https://github.com/Ultimaker/Cura/files/7850264/Booster.3mf.zip)

GregValiant, 2022-01-13:
Enable "Print Thin Walls" and set all the "Line Width" to 0.36 and make the "Outer Wall Inset" = 0.  I don't think "Retract Before Outer Walls" is going to help as they are all outer walls.  I think you can kick up your Accel and Jerk on this.  All the print moves are very short so Accel at 1000 and Jerk at 10 should work well.  The print time with 4.12.1 was 8:25 and with Arachne it was down to 3:08.  I'm not sure what's up with that.

The wall thicknesses (which look to be .39mm) of the honeycomb are right at the bottom of what is possible to print with a .4 nozzle.  That model would be much better with a 0.3 nozzle.
MS 3D Builder noted some problems but they were minor.  The unrepaired model sliced the same as a repaired one.
This is 4.12.1 with my suggested settings.  You can see that there are some yellow lines in there indicating very low flow.
[image]

This is sliced with Arachne Beta 2.  It took a bit of tweaking, but with a .4 nozzle I thought it looked pretty good.
[image]

fvrmr, 2022-01-13:
Hi @cfisher273 thank you for your report.
@GregValiant is correct! If you make your model bigger you it will slice correctly, so your wall are thinner than your nozzle size. 
You can check the Arachne build here: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta_2
I will close this issue, but please reach out if you have questions.

closed 2022-01-13 by fvrmr