Ultimaker/Cura #11248: Update profile with current settings/overrides. Greyed out and won't work

Ladnessness, 2022-01-12:
### Application Version

4.12.0 and 4.13.0

### Platform

macOS (Intel)

### Printer

Voron 2.4

### Reproduction steps

Quitting Cura and trying again.
Restarting computer
swap to ender 3 pro profile and back (ender 3 can update profile)
checked all plugins were enabled
Updated Cura to 4.13
changed between profiles while keeping settings and not keeping

### Actual results

No change button is still greyed out and won't perform action

### Expected results

Current settings should be saved to profile

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7851021/cura.log)
[image]

Ladnessness, 2022-01-13:
So I was running 4.12 and made some changes from the default settings in the profile. These changes (speeds) I wanted to be permanent for that profile. I am unable to save those changes to be the default settings for that particular profile. When I installed 4.13, to see if that would fix it, it replaced effectively replaces 4.12 on Mac. So on both versions after making changes it will not allow me to save those changes as defaults. Hope that explains it better

fvrmr, 2022-01-14:
Thank you for your explanation. You can't update default profiles. But you can create your own custom profile based on the default, by clicking on create profile from current settings/overrides... If you have a custom profile you can change the values and update the custom profile. 
I will close this issue, but let me know if it is unclear to you.

closed 2022-01-14 by fvrmr