Ultimaker/Cura #11250: Concentric infill should remain normal to build plate

johnco1126, 2022-01-12:
### Is your feature request related to a problem?

Concentric infill vertically follows walls when it should remain normal to the build plate.  Concentric infill that vertically follows walls will not support top surfaces if the walls are not normal to build plate.  The concentric infill will come to point if walls taper inward, or start floating if wall taper outward.
[image]

### Describe the solution you'd like

Change concentric infill to remain perpendicular to build plate. 

### Describe alternatives you've considered

Two concentric infill options. One vertically following the walls, and one that is perpendicular to build plate.

### Affected users and/or printers

Users that would prefer this particular type of infill. 

### Additional information & file uploads

[image]
[image]

fvrmr, 2022-01-14:
Hi @johnco1126 thank you for your request. 
Can you explain why this new type of concentric infill would be better than all the other infill patterns we already have.

johnco1126, 2022-01-14:
Some filaments to have issues with infill crossing over other infill lines on the same layer.  An infill that lays continuously without crossing would be helpful.  Either a concentric or a spiral infill would work.

fvrmr, 2022-01-18:
Hi @johnco1126 thank you for your reply. 
If you want an infill pattern that doesn't cross you can use other infill patterns like ZigZag and set the Infill line directions to one parameter or Gyroid or Cross. Read here more about different infill patterns: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/infill/infill_pattern.md

I have discussed this also with a Cura developer and we decided to reject this feature request. The reason why, is because when you are having a model that has different widths, the question is which walls should it follow for the models you showed above it could work but for ambiguous models it wouldn't. Since the walls could vary a lot and change, so that means that the concentric infill will not overlap properly. 
I hope this makes sense for you!

closed 2022-01-18 by fvrmr