Ultimaker/Cura #11252: Add an "explicit" option to Search and Replace

GregValiant, 2022-01-12:
### Is your feature request related to a problem?

When using a Layer Number as a search criteria then all instances of that search are replaced.  For example:
Search = ;LAYER:1
Replace = ;LAYER:1\nM117 Hello

The first result of that search in a gcode file of 1050 layers will be:
;LAYER:1
M117 Hello
That is as expected.  But Search and Replace goes on to replace any text string that starts with LAYER:1.
;LAYER:1, ;LAYER:10, LAYER:11, etc, and then all the LAYER:100 lines as well as all the LAYER:1000 lines.

My feature request is to add a checkbox and code to make the search string explicit so the replacements will only be made for exact matches.  In the specific case of layer numbers it could be done fairly simply by checking the string text and the string length(?)

### Describe the solution you'd like

When a situation arises that I want to make a replacement at ";LAYER:2" then I usually only want it to be for LAYER:2 and not LAYER:20, LAYER:200, et al.

### Describe alternatives you've considered

I wrote my own post-processor.  It certainly isn't as nice and clean as it would be from within Cura.

### Affected users and/or printers

All

### Additional information & file uploads

Thanks for your time.

fvrmr, 2022-01-13:
Hi @GregValiant thank you for your feature request. 
I will bring this up with the team. Keep you posted!

fieldOfView, 2022-01-13:
A current workaround is to search for `;LAYER:1\n`

GregValiant, 2022-01-13:
Thank You @fieldOfView.  With the Regex option enabled that works.
(I missed the Regex when I was playing around.)

closed 2022-01-13 by GregValiant

fvrmr, 2022-01-14:
Thank you @fieldOfView for the work around. 
Since there is a work around, I decided not to discuss this with the team as I estimated this to be low priority.