Ultimaker/Cura #11253: Repeated Slicing Errors

Dude-Industries, 2022-01-12:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Artillery Sidewinder X1

### Reproduction steps

I've been having problems getting a workable first layer with this one component, so started amending first layer height and line width; in desperation, I attempted a 0.15mm first layer; since then, the majority of attempts to slice have failed (even after resetting first layer to 0.2mm) 
[image]

### Actual results

repeated failure to slice :(

Have rebooted PC in hope of overcoming the gremlins, but they're clinging on with surprising tenacity

### Expected results

A wonderful and trouble free life enhancing experience?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I hope you can Cura my symptoms...

Dude-Industries, 2022-01-12:
Hold up, all sorted now... I also adjusted line width; instead of using percentage, I entered a value of 0.48 thinking it was in mm.

Reset to 100% and all is well; as a bonus, print speed is hugely improved!

closed 2022-01-12 by Dude-Industries

Ghostkeeper, 2022-01-19:
It shouldn't crash the slicing engine regardless. If a certain combination of settings causes this error we'd love to hear about it, because it could result in similar settings crashing for other people where they are useful.