Ultimaker/Cura #11254: Object moving off plate & will not print

onritter, 2022-01-12:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Lulzbot

### Reproduction steps

1. import STL file from inventor, object perfectly centered onto plate
2. slice it
3. put into Gcode file
4. object moves off plate into the corner where it cannot print
[image]

### Actual results

object is moved off plate into corner and cannot print

### Expected results

object remains in center of plate and is printed
[image]

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

GregValiant, 2022-01-12:
From the menu bar select "Settings" then "Printer" then down at the bottom "Manage Printers" and in that dialog select "Machine Settings".  On the left side are the dimensions of your build plate as X(width), Y(depth) and Z(height).  Make sure those numbers are correct.  There is a check box for "Origin at Center".  Make sure is is NOT selected (generally it is used by Delta printers).
Open a Gcode file with a text editor.  Near the beginning will be 6 numbers for the Max and Min of each axis for that print.  There should not be any negative values, and MaxX should be less than the width of your build surface and MaxY should be less than the depth of your build surface.

fvrmr, 2022-01-13:
Closing this issue since this is the Lulzbot Cura version and not our version. Please check if the suggestion of @GregValiant could help you. If not reach out the maintainer of that Cura version.

closed 2022-01-13 by fvrmr