Ultimaker/Cura #11257: PlugIn "Arc Welder" will not start / load after update Cura 4.13.0b -> 4.13.0

SpRiNgToIfEl, 2022-01-13:
### Application Version

4.13.0

### Platform

Win 10 Pro (with latest updates)

### Printer

Ender 5 Plus / BTT SKR PRO 1.2 / BTT TFT50v3 / TMC2209 UART

### Reproduction steps

Update Cura 4.13.0b to 4.13.0

1. Execute "Ultimaker_Cura-4.13.0-amd64.exe", check "deinstall old version" / "keep setting"
2. Start Cura, the PlugIn will not start / loaded.
3. activate the check-button, restart cura, same issue

4. deinstall PlugIn, reinstall PlugIn
5. Same issue as above

### Actual results

The PlugIn will not starts / works with Cura 4.13.0

### Expected results

Does anyone have the same experience?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]
[image]

fieldOfView, 2022-01-13:
Please report issues with the ArcWelder plugin in the issuequeue for that plugin:

https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues

If you do, please include the Cura.log logfile, which can be found via the Help -> Show configuration folder.

SpRiNgToIfEl, 2022-01-13:
unfortunately there is no cura.log in the folder

fieldOfView, 2022-01-13:
The file is most certainly there, but Windows may be hiding the .log extension. In that case the file will just show up as "Cura". There may be multiple files just named "Cura" (the other being Cura.cfg".

fieldOfView, 2022-01-13:
Please continue this issue here: https://github.com/fieldOfView/Cura-ArcWelderPlugin/issues/26