Ultimaker/Cura #11258: Tree Support collides with Object

slewantoski, 2022-01-13:
### Application Version

4.13.0

### Platform

Windows 11

### Printer

Creality CR-10S Pro

### Reproduction steps

1. import STL
2. enable Tree Support
3. Preview

### Actual results

Until Version 4.12.1 Tree Support worked fine.
Since 4.13.0 the Tree Support collides with the object and can not be removed after printing

I've attached an example file where the problem occurs.

### Expected results

Tree Support should not be printed in the object.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[cura-log.zip](https://github.com/Ultimaker/Cura/files/7862572/cura-log.zip)
[Project.zip](https://github.com/Ultimaker/Cura/files/7862573/Project.zip)
[image]
[image]

fvrmr, 2022-01-13:
Hi @slewantoski thank you for your report.
We have a ticket on our backlog for this issue. I will link your issue to our ticket!
Devs see CURA-8852

ajalberd, 2022-01-13:
I can confirm this is an issue with hollow models printed at an angle - the tree supports cut through and merge with the walls making them impossible to remove.

This is not an issue in the 4.13 beta, at least for me.

partlyhuman, 2022-01-14:
I have some more repro cases for this issue. Attaching a test project.
[image]
[issue-test-project.zip](https://github.com/Ultimaker/Cura/files/7866734/issue-test-project.zip)

chipmunk1977, 2022-01-20:
Is it possible that this is somehow a Windows issue?  I've tried several models and various settings within tree supports and can't reproduce this at all on a Linux machine.

Ghostkeeper, 2022-01-20:
I wouldn't think this sort of issue is Windows-specific (but indeed so far all reports for this issue are on Windows). Operating-system-specific issues do exist in CuraEngine, but are pretty rare. In this case it should at the very least remove the intersection of support interface with the model.

GregValiant, 2022-01-22:
Here is one from the Ultimaker Forum.  Windows 11 and Cura 4.13.
[Issues with Tree Supports](https://community.ultimaker.com/topic/39277-issues-with-tree-supports-on-ender-6-cura-413/#comment-301810)

mjquinlan2000, 2022-01-23:
I can confirm that tree supports are colliding with the model on macOS 12.1

nallath, 2022-01-24:
This has been fixed. We will release a 4.13.1 ASAP. Thanks for reporting it :)

closed 2022-01-24 by nallath