Ultimaker/Cura #11259: Extra Plane when Slicing Parts

zacharyneveu, 2022-01-13:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

1. Open the attached project
[bugged_cura_slice.zip](https://github.com/Ultimaker/Cura/files/7863576/bugged_cura_slice.zip)

2. Slice in default orientation (see photo below) without supports (bug also appears with supports)
[image]

### Actual results

Slice shows 2 extra planes being printed in places where the STL had no material (see photo below)
[image]

If the part is rotated by a small amount the planes do not appear. If the part is rotated 180 degrees (upside down), the planes appear again.
[image]

### Expected results

No extra planes should appear. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[bugged_cura_slice.zip](https://github.com/Ultimaker/Cura/files/7863610/bugged_cura_slice.zip)
[cura_bugged_slice.log](https://github.com/Ultimaker/Cura/files/7863611/cura_bugged_slice.log)

fvrmr, 2022-01-14:
Hi @zacharyneveu thank you for your report. 
Looking at your model, your model is not correct. You can red on top, which means that area need support. So the normals of that model are reversed. If you install the [mesh tools plugin](https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools) you can fix your model normals. 

[image]
[image]

I will close this issue, but reach out if you have more questions.

closed 2022-01-14 by fvrmr

zacharyneveu, 2022-01-14:
Sorry for the spam and appreciate the helpful advice, that solved it!