Ultimaker/Cura #11261: Travel after Part printed in sequence will hit other Part

Malaber, 2022-01-14:
### Application Version

4.12.1

### Platform

Windows 10

### Printer

Ender 5 Plus

### Reproduction steps

1. I placed my parts spread out on the build plate to print them in sequence
2. Cura does not provide a Setting on print order (print this part, than this one and so on)
3. Cura generates a travel path between Parts which will collide with the previously printed Parts

I attached my Cura saved Project: [Ski_Helm_Mask_Holder_Set_PETG.zip](https://github.com/Ultimaker/Cura/files/7869338/Ski_Helm_Mask_Holder_Set_PETG.zip)
 
This is the path that will interfere with a printed part:
[image]

Ideas to fix it:

- When printing in "One at a time" mode print from smallest (z axis) object to largest; this way the nozzle cannot interfere with previously printed parts.
- Move around previously printed parts instead of just going the fastest route
- Provide a print order setting for the user to manually check this (tedious)

### Actual results

Cura generates a travel path between Parts which will collide with the previously printed Parts.
[image]

### Expected results

Cura generates a travel path between Parts which will **NOT** collide with the previously printed Parts.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Ski_Helm_Mask_Holder_Set_PETG.zip](https://github.com/Ultimaker/Cura/files/7869338/Ski_Helm_Mask_Holder_Set_PETG.zip)
[Colliding travel path](https://user-images.githubusercontent.com/32635600/149492453-bdd99883-84e8-421a-b5ad-8204be93fb61.png)

fvrmr, 2022-01-14:
Hi @Malaber thank you for your report. 
If you import your gcode you don't see the travel moves will not hit printed parts. Or did you print this project file and did it hit the other models?

About the order see this: https://github.com/Ultimaker/Cura/issues/3070

Malaber, 2022-01-17:
[CE5P_Ski_Helm_Mask_Holder_Set_PETG.zip](https://github.com/Ultimaker/Cura/files/7884289/CE5P_Ski_Helm_Mask_Holder_Set_PETG.zip)
Here is the GCODE it generated. Looking at it now, I think it would not have hit the print. 
I paused the print and removed the part before just to be sure. Still it is a bug in the preview pane after slicing I think.

fvrmr, 2022-01-18:
Thank you for your reply. I will bring this up with the team. Keep you posted!

fvrmr, 2022-01-21:
I have discussed it with the team and decided to defer this issue, since this has not a high priority for us to fix. 
I hope you understand.

closed 2023-06-10 by github-actions[bot]