Ultimaker/Cura #11262: Menus are broken in Ubuntu

ausbradr, 2022-01-14:
### Application Version

4.13

### Platform

Ubuntu 20.04

### Printer

Ender 3 Pro, but that is irrelevant.

### Reproduction steps

1. Open Cura, try clicking on menus
2. Observe how the menus won't stay open long enough for you to click items, or will open up elsewhere on the screen if you're lucky. 
3. Use the menus too much, and the app locks up. 

### Actual results

I can't get my work done, lol

### Expected results

We should be able to use the menu bar as intended.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

ausbradr, 2022-01-18:
> Hi @ausbradr <https://github.com/ausbradr> thank you for your report.
> Are you using our AppImage?

It was installed using snap.

fvrmr, 2022-01-18:
We don't maintain the snap packages only the AppImage. You can find our AppImage here; https://github.com/Ultimaker/Cura/releases/tag/4.13.0
Please report your issue to the maintainers of the Snap packages.

I will close this issue since we won't fix this.

closed 2022-01-18 by fvrmr