Ultimaker/Cura #11264: RRF Fan Speed Bug

cr0wm4n, 2022-01-15:
### Application Version

>4.10.0 

### Platform

Windows 10

### Printer

RRF

### Reproduction steps

generate RRF gcode using regular fan speed 0%
Maximum fan speed 5%

### Actual results

Fan blows crazy speeds at times

### Expected results

RRF can use 2 Fan speed scales
0.0-1.0 (with decimal place)  and 0-255 (without decimal place)

cura generates M106 commands like below using the low flow settings I am using and appears to be a combination of both of RRF scales.

Please see the lines below pointed at with Arrows
Cura needs to use either the 0.0-1.0 scale or 0-255 scale not a combination of both, 
The resulting fan on the highlighted lines below destroyed my print with expensive material

	Line 994075: M106 S3.7
	Line 994455: M106 S2.3
	Line 994837: M106 S0.6    <--- RRF sees this as 60%
	Line 995228: M106 S1.4
	Line 995624: M106 S3.7
	Line 996005: M106 S2
	Line 996371: M106 S2.3
	Line 997099: M106 S3.3
	Line 997808: M106 S4.2
	Line 998159: M106 S3.6
	Line 998821: M106 S3
	Line 1000806: M106 S0.3    <--- RRF sees this as 30%
	Line 1001142: M106 S5.7
	Line 1001448: M106 S5.4
	Line 1001747: M106 S5.9
	Line 1002053: M106 S5
	Line 1002377: M106 S4.5
	Line 1002987: M106 S1.7
	Line 1003289: M106 S2.1
	Line 1003581: M106 S3
	Line 1003856: M106 S5.6
	Line 1004127: M106 S3.1
	Line 1004404: M106 S3.4
	Line 1004678: M106 S3.7
	Line 1005227: M106 S4.1

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[CFFFP_RR_motormount.zip](https://github.com/Ultimaker/Cura/files/7873868/CFFFP_RR_motormount.zip)

fvrmr, 2022-01-18:
Hi @cr0wm4n thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

GregValiant, 2022-01-19:
I don't think it's a combination of scales.  Cura rounds the fan speeds to 1 decimal place so 5% of 255 = 12.8 and that would be in the M106 line and the display would read correctly (as opposed to entering 70% and seeing 69% on the LCD if you truncate to an integer).  From MarlinFW.org the S parameter of M106 is defined as:
**"[S (Speed)]       Speed, from 0 to 255. S255 provides 100% duty cycle; S128 produces 50%."**

It would seem that there is just PWM scale and no implementation of some alternate scale from 0.0 to 1.0.
In those later lines in the gcode file the fan speed is being adjusted by the bridge settings.  As you have Cura set up, your "Regular" fan speed is 0 and your Maximum fan speed is 5%.  When the Bridge setting makes an adjustment it causes those really small numbers that your firmware appears to be misinterpreting.  If you set the Regular fan speed equal to the Maximum fan speed then the issue does not occur as the bridge settings won't affect the fan.  Essentially the fan will run at 5% from gcode Layer:3 to the end of the print.
It would seem that since they provide their own software (and firmware?) that they might have a command to tell the firmware to use EITHER PWM scale OR their proprietary fan speed scale, but not both at the same time.  I would ask Markforge about that as it may have come up before in regards to their Eiger software.

cr0wm4n, 2022-02-04:
> I don't think it's a combination of scales. Cura rounds the fan speeds to 1 decimal place so 5% of 255 = 12.8 and that would be in the M106 line and the display would read correctly (as opposed to [...]

This is an issue as the pwm scale should be whole numbers 0-255 no decimal points, decimal points are used on the 0.0 - 1.0 scale.

cr0wm4n, 2022-02-04:
thankyou @rburema this should resolve the issue :)

commit 35045170 (2022-01-26): Add 'Scale Fan Speed From 0 to 1' Setting.

Will help fix #11264 -- See also backend (CuraEngine) PR with the same branch-name.
PR #11345 (2022-03-25): Add 'Scale Fan Speed From 0 to 1' Setting.
Will help fix #11264 

See also backend (CuraEngine) PR with the same branch-name: https://github.com/Ultimaker/CuraEngine/pull/1564
closed 2022-03-25 by rburema