Ultimaker/Cura #11265: Machine Settings button missing

mattpenner, 2022-01-15:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Creality Ender 5

### Reproduction steps

Install Cura 4.13.0 (replaced 4.11.0).

Go to Settings - Printers - Manage Printers

No Machine Settings button is shown.

When attempting to select the printer and click Activate there is no change.

### Actual results

No Machine Settings button is shown.

### Expected results

When the active printer is selected the Machine Settings button should be available.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

mattpenner, 2022-01-15:
I apologize.  This looks to have been an issue with my profiles.  I installed both 4.12.1 and back to 4.11.0 and they both exhibited the same issue.  I backed up my profiles, deleted my profile directories using Option 2 on this guide and restarted and now the buttons are back.  After restoring my profiles all is well.

[image]

closed 2022-01-15 by mattpenner

Ghostkeeper, 2022-01-21:
The machine settings action is a plug-in that could've been disabled. The buttons shown there is also a list in the printer definition file if you had changed those.