Ultimaker/Cura #11266: "None" build plate adhesion option doesn't make entire building area available

middlenamefrank, 2022-01-15:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

Creality CR-10S

### Reproduction steps

Set build plate adhesion option to "None"

### Actual results

Area is still reserved for a skirt/brim.

### Expected results

Entire build plate area should be made available, as it is with a "Brim" option of 0 width. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

GregValiant, 2022-01-17:
I don't have 4.12.0 but in 4.12.1 I cannot reproduce this.  Looking in the definition files there are no "disallowed areas".
If you zip a 3mf project file ("File / Save Project") and post it maybe it will show something.  If you go to "Help / Show Configuration Folder" and zip the Cura.log file and post that here as well it might show something.
[image]

Ghostkeeper, 2022-01-21:
The build plate adhesion is not the only thing that affects the border. Another prominent feature is Avoid Printed Parts When Travelling and its accompanying avoid distance, as well as the used extruders and the nozzle offset. Less common features are the draft shield and ooze shield.

The full list of settings affecting the disallowed areas is:

```
adhesion_type
brim_line_count
brim_width
draft_shield_dist
draft_shield_enabled
extruder_prime_pos_x
extruder_prime_pos_y
infill_wipe_dist
initial_layer_line_width_factor
machine_nozzle_offset_x
machine_nozzle_offset_y
material_shrinkage_percentage
material_shrinkage_percentage_xy
material_shrinkage_percentage_z
ooze_shield_dist
ooze_shield_enabled
prime_blob_enable
prime_tower_brim_enable
prime_tower_enable
prime_tower_position_x
prime_tower_position_y
prime_tower_size
raft_base_extruder_nr
raft_interface_extruder_nr
raft_margin
raft_surface_extruder_nr
skirt_brim_extruder_nr
skirt_brim_line_width
skirt_gap
skirt_line_count
support_bottom_enable
support_bottom_extruder_nr
support_enable
support_extruder_nr_layer_0
support_infill_extruder_nr
support_offset
support_roof_enable
support_roof_extruder_nr
travel_avoid_distance
travel_avoid_other_parts
travel_avoid_supports
wall_line_count
wall_line_width_0
wall_line_width_x
```

closed 2022-01-31 by no-response[bot]