Ultimaker/Cura #11267: include Creality CR6MAX

marcelrios, 2022-01-15:
### Application Version

4.13.0

### Platform

MAC OS

### Printer

Creality CR6 Max

### Reproduction steps

I need you to include the configuration of the Creality CR6 MAX

### Actual results

I need you to include the configuration of the Creality CR6 MAX

### Expected results

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

.

GregValiant, 2022-01-15:
Creality (or trouch.com) provides the printer definition files for their printers.  Until they get around to it you can install the CR6-SE, change the name in the right side of the dialog box, and then change the build volume (Xwidth, Ydepth, Zheight) in the machine settings when it comes up.

fvrmr, 2022-01-17:
Closing this issue, since we don't add third party printers we are depended on our community to contribute these profiles to us.

closed 2022-01-17 by fvrmr