Ultimaker/Cura #11269: Notifactions and warnings don't disappear

BSOD76, 2022-01-15:
### Application Version

4.9.8 ->

### Platform

Windows 10

### Printer

ender3 pro

### Reproduction steps

1. save file, upload to printer or anything that evokes a pop-up

### Actual results

[image]

### Expected results

pop-up should disappear after 30 sec.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

An option would be nice to disable pop-ups altogether.

fvrmr, 2022-01-17:
Hi @BSOD76 thank you for your report.
Does this only happen with these notifications when sending to the printer? 
I assume you are using a plugin to send your print jobs to the printer right?

fvrmr, 2022-01-18:
Closing this issue since it is indeed related to the Creality Cloud Integration plugin, thanks @fieldOfView!
We can't fix this since we don't maintain this plugin. Please report it to the maintainers of this plugin.

closed 2022-01-18 by fvrmr