Ultimaker/Cura #11270: Textposition in Project Save Windows (GERMAN lang.)

soundserie, 2022-01-15:
### Application Version

4.13.0

### Platform

MACOS 12.1 

### Printer

not relevant

### Reproduction steps

Project after slice process save.

### Actual results

German text ist in down layer ant buttons cover this text.
[image]

### Expected results

Text is complet readable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

not relevant

fvrmr, 2022-01-17:
Hi @soundserie thank you for your report.
I will bring it up with the team. Keep you posted.

fvrmr, 2022-01-18:
I have discussed it with the team and it turns out that this is a known issue. It turns out you can resize your window to show the text. And we decided to defer this, since this will change with some updates we have planned. I hope you understand.

closed 2023-06-10 by github-actions[bot]