Ultimaker/Cura #11271: The number of wall lines is always more than one. I want only one wall line

nick2mail, 2022-01-15:
### Application Version

4.13.0

### Platform

Win10

### Printer

Custom (FlSun Cube)

### Reproduction steps

nozzle -0.8. Layer height - 0.6. extrusion width any from 0.6 to 1.5.  wall thickness any from 0.6 to 1.5. 
I set the number of wall lines = 1 !!!   
Problem  -I get the real slicing mode =2 - why is that?
[image]
[Documents.zip](https://github.com/Ultimaker/Cura/files/7876030/Documents.zip)

### Actual results

Problem  -I get the real slicing mode =2 - why is that?

### Expected results

other slicers  can make a single wall layer

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[CFFFP_Bottom_box.zip](https://github.com/Ultimaker/Cura/files/7876034/CFFFP_Bottom_box.zip)

GregValiant, 2022-01-16:
The walls are 1.5mm thick and your Line Width is 1.0mm.  There is air on both sides and so there is an inside "outer wall" and an "outside" outer wall for a total of 2 walls.  If you change the line width to 1.49mm you will see that the upper portion consists of a single wall.  The lower "sill" will still be 2 walls wide as it is 2.0mm thick.  If you want a model with a single wall you would need to design it with 1.0mm walls everywhere.  Cura is slicing the model you provided.

On a model with fine details like that one has you should consider switching to a .4 nozzle as the .8 has trouble printing fine details.  It's only a 27 minute print.
This is with PrusaSlicer and there are two walls.
[image]

fvrmr, 2022-01-18:
Hi @nick2mail thank you for your report.
@GregValiant explanation is correct. This is simply how Cura works with your model. I will close this issue, but reach out if you have more questions.

closed 2022-01-18 by fvrmr