Ultimaker/Cura #11272: Monitor panel not updated switching 3D printer

giufini, 2022-01-16:
### Application Version

4.13.0

### Platform

Windows 10

### Printer

Elegoo Neptune 2S

### Reproduction steps

1. Add and configure two 3D printers equipped with WiFi module
2. Connect to a WiFi 3D printer and open the Cura MONITOR panel
3. Select a different 3D printer from the drop-down menu

### Actual results

The 3D printer info in the Cura MONITOR panel are not updated for the new selected printer

### Expected results

The 3D printer info in the Cura MONITOR panel should be updated for the new selected printer

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fieldOfView, 2022-01-16:
Out of the box, Cura connects to printers over USB (which is not really being supported by Ultimaker, because no Ultimaker printers can print over USB), or via the proprietary Ultimaker protocols. If you connect to your (non Ultimaker) printer via Wifi, you must be using a plugin. If so, please log an issue with the creators of that plugin.

fvrmr, 2022-01-17:
Hi @giufini are you using the Ultimaker Cura version or the Elegoo Cura version?
We can only help you with the Ultimaker Cura version.

Ghostkeeper, 2022-01-24:
Elegoo Cura is still on version 4.8, so we can be certain this is with Ultimaker Cura.

Ghostkeeper, 2022-01-24:
Elegoo printers are connected via the MKS Wifi plug-in, which is not maintained by us. Please report this issue here: https://github.com/Jeredian/mks-wifi-plugin/issues

closed 2022-01-24 by Ghostkeeper