Ultimaker/Cura #11274: Strange preview result (like sunrays everywhere)

altimac, 2022-01-16:
### Application Version

4.12.1

### Platform

MacOS

### Printer

FLSun Speed Race

### Reproduction steps

Open the given STL
Slice
Preview

### Actual results

[image]

### Expected results

A much more normal preview

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Here is the STL file [Mac_SE30_Nano_Dock_-_Thingiverse.stl.zip](https://github.com/Ultimaker/Cura/files/7876877/Mac_SE30_Nano_Dock_-_Thingiverse.stl.zip)

GregValiant, 2022-01-16:
Nice!
I don't get any of that.  I'm thinking it's an error in the visual representation.  If you create a Gcode from that slice, clear the build plate, and then open the Gcode in Cura how does it look?  Does it show those "Rainbow" moves?

altimac, 2022-01-16:
I do not know how to create gcode from a slice, but I quit Cura and reopen de same STL file... and it worked as expected 🤷‍♂️

altimac, 2022-01-16:
Note that I already had this bug with another STL file few days ago.

GregValiant, 2022-01-16:
After you slice the button will say "Save to Disk".  When you do that a gcode file will be created.  Just open it in Cura.
It does sound like a video system problem though.  Maybe you can update the video drivers?

fvrmr, 2022-01-17:
Hi @altimac thank you for your report.
This is a duplicate of https://github.com/Ultimaker/Cura/issues/10387 so I will close this issue. 
Please continue the thread there.

closed 2022-01-17 by fvrmr