Ultimaker/Cura #11275: Improving notifications

abelbour, 2022-01-16:
### Is your feature request related to a problem?

I find a bit annoying having to dismiss every notification so they don't end up covering the main area of the window.

### Describe the solution you'd like

I think there are many ways to improve on this, maybe a notification area that doesn´t cover the main window. 

### Describe alternatives you've considered

A heads-up notification or toast that dismisses automatically while a notification center shows all the previous notification

### Affected users and/or printers

It will improve the workflow for everyone by not having to dismiss notifications constantly.

fvrmr, 2022-01-17:
Hi @abelbour thank you for your request.
Could you explain a bit more what notifications you are getting? With some notifications you will get asked to not show up again.

abelbour, 2022-01-17:
[image]
Thanks for answering, I'm sorry I hadn't made myself clear.
The notification area, being centered in the workspace, covers it pretty quickly if you don't clear them often, which I find a bit cumbersome, as seen in the attached image.
I hope this helps improving an already great slicer. Thanks again!

fvrmr, 2022-01-18:
Thank you for your screenshot. Looking at this it is a duplicate of https://github.com/Ultimaker/Cura/issues/11269
These notifications are from the Creality Cloud Integration plugin, which we don't maintain. Please report this to the maintainers of that plugin. 
I will close this issue since it is a duplicate.

closed 2022-01-18 by fvrmr

abelbour, 2022-01-18:
Actually they're from the Moonraker plugin, but that's not the point, it's the same with a lot of notifications, even from Cura itself. The problem is the positioning of them in the interface. I hope this helps, thank you.

fieldOfView, 2022-01-18:
The notifications from Cura itself can be fixed by the Cura devs. If you have a notification from Cura itself that does not go away automatically, please report that specific notification message.

Ghostkeeper, 2022-01-24:
Most notifications should disappear after 30 seconds automatically. There's a few that are infinite though, for which it was deemed to be critical that the user interacts with them or does something to fix them. Those should be kept to a minimum and be removed when they no longer apply.