Ultimaker/Cura #11276: Project name reverts to previous project name after saving

mike5577, 2022-01-16:
### Application Version

4.13.0

### Platform

Windows 10 21H2

### Printer

Ender 3 v2

### Reproduction steps

1. Open a model
2. Slice and save gcode to disk
3. Save project as 3mf
4. Clear build plate
5. Open new model
6. Slice and save gcode to disk
7. Project name at this point will have reverted to the name of the first project
8. Go to save project as 3mf and it will try to overwrite the last project since they now have the same name

### Actual results

After step 6, the project name reverts to whatever the last project was called. I believe this bug started with 4.12.0.

### Expected results

Project name should not change after saving gcode, should stay the same. Since I didn't modify the project name, it should just be whatever the prefix is for my printer, followed by the filename for the STL I loaded.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Timestamp in log for replicating the bug should be around 11:28 or slightly before.
[cura - Copy.log](https://github.com/Ultimaker/Cura/files/7877676/cura.-.Copy.log)
https://user-images.githubusercontent.com/97849493/149673470-b7f5a784-a660-4f03-b4b7-df00b01644b1.mp4

Ghostkeeper, 2022-01-16:
Since 4.12 it should change the print job name to the filename the project file was saved at when saving a project file.

However it shouldn't change the print job name when saving .gcode files, and certainly not to something else than the filename of that .gcode file!

I'm guessing the file name got stored from when the project file was saved and didn't get overwritten from memory when the toolpath was saved, but saving the toolpath did (erroneously) restore the name from that piece of memory.

fvrmr, 2022-01-17:
Hi @mike5577 thank you for your report.
I will bring it up with the team, keep you posted.

fvrmr, 2022-01-18:
I have discussed it with the team and we created a ticket on our backlog for this.
Devs see CURA-8897

GregValiant, 2022-02-07:
I've been having this problem with Project Names in 4.13.1.  It reproduces like this:
I open an STL
Save the Gcode and then Save Project as MyProject.3mf
Clear the build plate
Open another file (project or STL)
Save as Project as Dork.3mf - it properly saves.
Immediately after the Save dialog closes the Project Name reverts to the previous project name MyProject.

fvrmr, 2022-02-10:
@GregValiant thank you, I could not always reproduce this. But it happened for me at least twice. I will add this to our ticket.

GregValiant, 2022-02-10:
My pleasure.  This one can be truly annoying.

GregValiant, 2022-05-04:
@fvrmr -This problem has bitten me 3 or 4 times in the last couple of days.  I really hope it gets addressed in the next release.

GregValiant, 2022-05-20:
This problem still occurs in 5.0.0.

GregValiant, 2022-07-28:
The problem still exists in 5.1.0.
This is a productivity issue.  Over-writing files causes users to have to repeat all their work to recreate the gcode for the correct model.

adobni, 2023-09-27:
Still happening in Cura 5.4.0. This is very frustrating. I have accidentally overwritten projects because the name changes unexpectedly.

GregValiant, 2023-09-28:
I've noticed.  It jumped up and bit me twice this morning and it isn't 9' o'clock yet.

ucimo, 2024-01-18:
Still happening on linux

GregValiant, 2024-03-26:
5.7 is still a problem.  If the gcode is saved with a name other than the project name, the project name reverts to a previous project.

Robbi62, 2025-01-23:
This critical issue is still alive in Cura 5.9.0.

As I didn't find this bug report before, I created a new issue #20175 with a more detailled description.

This issue destroys project files from a former project, you worked on even days before!