Ultimaker/Cura #11279: When saving a large file (eg. takes 10sec to save) if you close the program (x) it will corrupt the file

kylane, 2022-01-17:
### Application Version

4.12.1 & 4.13.0-BETA

### Platform

Win10

### Printer

Creality Ender 3

### Reproduction steps

Opened a large 3mf file, made edits, saved it - as it was saving (didnt realise it was still saving) hit the close app X in the top right and app closed. Went to open file later, and it was corrupt.

### Actual results

Saved file was corrupt.

### Expected results

From a UX POV : perhaps a dialog should appear before closing the app while files are being saved?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No need to add files.

fvrmr, 2022-01-17:
Hi @kylane thank you for your report.
When a file is saving it is showing a pop up to show it is saving your project. 
But I can reproduce your issue. I will bring it up with the team. Keep you posted.

kylane, 2022-01-17:
> 

HI there - it doesnt always show the popup dialog when saving - sometimes it may appear for a quick second at the end of saving .

To reproduce, I open my large file - make some edits - send it to octoprint then save - I dont get a "saving" dialog until the very end of saving.

nubnubbud, 2022-01-18:
this seems like an easy fix, but there are multiple solutions. is it better to just keep it running or open a dialogue when saving, or should it make a second file temporarily, then switch the files' names and transfer metadata before deleting the original? the second option here would also work in the case of power loss or crashes or similar.

fvrmr, 2022-01-18:
I could reproduce your issue! And so I have discussed it with the team and created a ticket for this on our backlog. 
Devs see CURA-8898

kylane, 2022-01-18:
> I could reproduce your issue! And so I have discussed it with the team and created a ticket for this on our backlog. Devs see CURA-8898

Nice one! Good luck!