Ultimaker/Cura #11282: Cura  doesn't start up properly after update

OhaDerErste, 2022-01-17:
### Application Version

4.13

### Platform

Arch Linux, My GPU is a amd radeon rx 550

### Printer

Ender 3 v2

### Reproduction steps

1. Update Cura from 4.12 to 4.13 with pacman
2. Open Cura

### Actual results

[image]
I get a lot of errors.
If i click on remove plugin, cura crashes

### Expected results

Cura should start up normally

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Log:
```
Thread 0x00007fcf1f6fe640 (most recent call first):
  File "/usr/lib/cura/plugins/USBPrinting/USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf1feff640 (most recent call first):
  File "/usr/lib/cura/plugins/RemovableDriveOutputDevice/RemovableDrivePlugin.py", line 61 in _updateThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf35ffb640 (most recent call first):
  File "/usr/lib/python3.10/site-packages/UM/Backend/Backend.py", line 165 in _storeStderrToLogThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf367fc640 (most recent call first):
  File "/usr/lib/python3.10/site-packages/UM/Backend/Backend.py", line 153 in _storeOutputToLogThread
  File "/usr/lib/python3.10/threading.py", line 946 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf57fff640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf78ff9640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf797fa640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf79ffb640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf7a7fc640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf7affd640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf7b7fe640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Thread 0x00007fcf7bfff640 (most recent call first):
  File "/usr/lib/python3.10/threading.py", line 320 in wait
  File "/usr/lib/python3.10/threading.py", line 460 in acquire
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 98 in _nextJob
  File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 124 in run
  File "/usr/lib/python3.10/threading.py", line 1009 in _bootstrap_inner
  File "/usr/lib/python3.10/threading.py", line 966 in _bootstrap

Current thread 0x00007fcfdecba740 (most recent call first):
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 396 in _logInfoWidget
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 182 in _createDialog
  File "/usr/lib/python3.10/site-packages/cura/CrashHandler.py", line 86 in __init__
  File "/usr/bin/cura", line 162 in exceptHook
  File "/usr/lib/python3.10/site-packages/UM/Message.py", line 109 in _onActionTriggered
  File "/usr/lib/python3.10/site-packages/UM/Signal.py", line 332 in __performEmit
  File "/usr/lib/python3.10/site-packages/UM/Signal.py", line 219 in emit
  File "/usr/lib/python3.10/site-packages/UM/Qt/Bindings/VisibleMessagesModel.py", line 105 in actionTriggered
  File "/usr/lib/python3.10/site-packages/UM/Qt/Bindings/MainWindow.py", line 203 in mouseReleaseEvent
  File "/usr/lib/python3.10/site-packages/UM/Qt/QtApplication.py", line 396 in exec_
  File "/usr/lib/python3.10/site-packages/cura/CuraApplication.py", line 877 in run
  File "/usr/bin/cura", line 236 in <module>

```
Error traceback
```
Traceback (most recent call last):
  File "/usr/lib/python3.10/site-packages/UM/PluginRegistry.py", line 505, in <lambda>
    unable_to_load_plugin_message.pyQtActionTriggered.connect(lambda message, action: self._acceptedRemoveCorruptedPluginMessage(plugin_id, message))
  File "/usr/lib/python3.10/site-packages/UM/PluginRegistry.py", line 481, in _acceptedRemoveCorruptedPluginMessage
    message_data = self.uninstallPlugin(plugin_id)
  File "/usr/lib/python3.10/site-packages/UM/PluginRegistry.py", line 537, in uninstallPlugin
    "message": message_text,
UnboundLocalError: local variable 'message_text' referenced before assignment

```

fieldOfView, 2022-01-17:
https://github.com/Ultimaker/Uranium/blob/4.13/UM/PluginRegistry.py#L512 is a bit iffy. Things seem to happen twice, and the message that is returned is not defined unless the plugin is removed from the Marketplace dialog.

closed 2022-01-18 by fvrmr

fieldOfView, 2022-01-18:
@fvrmr the Remove plugin button crash still needs investigation IMHO.

fvrmr, 2022-01-18:
@fieldOfView I had discussed it with @Ghostkeeper, turns out this only happens when the uninstallPlugin is called external but I will discuss this with the team.

fvrmr, 2022-01-21:
I have discussed it with the team and created a ticket for the uninstallPlugin. Devs see CURA-8905