Ultimaker/Cura #11283: Flsun QQ wifi communication problem with Cura

michelconf, 2022-01-17:
### Application Version

4.13.0

### Platform

Windows 8.1 Pro

### Printer

Flsun QQ

### Reproduction steps

1. Connect my printer to the network - IP 192.168.1.104
2. Open Cura
3. Printer connection in first step
4. Loosing the connection to the printer after one or two second
5. Note that I succeed one time to be connected and print. Since that I am not able to stay connected for one or two second.

### Actual results

Printer not connecte
[image]
d

### Expected results

Printer connected and operating.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Nothing more to say or to add.

michelconf, 2022-01-17:
[image]

michelconf, 2022-01-17:
Printer always ready and connected to my network

closed 2022-01-17 by michelconf

Ghostkeeper, 2022-01-24:
Please report this with whichever plug-in maintains the connection with your firmware! It's not in the default installation of Ultimaker Cura, so we can't fix this.